KenT2/pipresents-gapless #88: Can't edit Track - Web Editor error

aroomthedoomed, 2018-09-21:
Hi Ken,

I've got this error when trying to click on OK when editing a track in a radiobuttonshow

`
remi.server.ws   ERROR    error parsing websocket
Traceback (most recent call last):
  File "/home/pi/pipresents/remi/server.py", line 262, in on_message
    callback(**param_dict)
  File "/home/pi/pipresents/pp_web_edititem.py", line 234, in confirm_dialog
    self.field_content[field]=str(self.field_objs[field_index].get_value()).strip()
UnicodeEncodeError: 'ascii' codec can't encode characters in position 13-15: ordinal not in range(128)
`

So I can't edit the video track anymore. This happen after changing some command control in a show.

Cheers

KenT2, 2018-09-21:
The error message indicates that you typed a non ASCII character. ASCII are basically the Latin alphabet, numbers, and some punctuation.
I don't think the edited track will have been saved to disc but if the problem persists you may need too edit the appropriate .json file in the profile directory with a text editor to remove the offending character.
PP does not support unicode which are characters other than ASCII.

aroomthedoomed, 2018-09-22:
I investigated the .json file, ans I think the issue came from the media name, who contain a space. changing the name from xxx_800 x 600.mp4 to xxx_800x600.mp4 seems to resolve the issue.

aroomthedoomed, 2018-09-24:
after further investigation, the problem wasn't with the space, but with the x. I don't know why, but the character x was change for a × (multiplication sign), who caused the problem.

KenT2, 2018-09-26:
multiply is not ASCII so would not work. The question is, how did it get there, maybe one of the browsers knew better than you !

closed 2018-12-13 by KenT2