KenT2/pipresents-gapless #89: Animation support in Liveshows

aboomer, 2018-09-24:
I have a Radiobuttonshow that contains four tracks, each of which are a liveshow.  My aim is to control one of 4 LEDs to indicate to a user which liveshow is currently active.

In the "Animation Control" section 13 of the manual it discusses adding animation control to tracks, however I don't see a way to apply these control to my liveshows (given that they are not actually tracks as such).  Can any one help?

Thanks,
Alastair

KenT2, 2018-09-26:
There is no simple way to do it. What you need to do is:
Create four single run shows each with a single audio track that has no file to play and zero duration. Use the track to control the LED
In each liveshow use Show Control to Open the single run show at the start of the show and close it at the end.
I haven't tried it. If it does not work then I think I have another solution which involves me in a bit of coding - to add a Show Control command that does animation.

aboomer, 2018-09-26:
Hi Ken,

Thank you for the guidance, that sounds like it should work.  After giving this more thought I'm considering going down the route of OSC control instead.  In my application I'm planning to put a TV in and out of standby (via an IR blaster) and confirm that it has done so by monitoring its supply current (via a CT sensor).  I'm thinking it therefore would be better for me to control everything from a single state-machine to determine how GPIOs etc should be processed depending on context.  This state-machine would then control Pi Presents (as appropriate) via OSC control.  Although this might be a more complicated method overall, it should give me good control over the entire system and also help me gain some learnings along the way!

Thanks,
Alastair

P.S. The Pi Presents software that you've put together is great!

closed 2019-04-15 by KenT2