KenT2/pipresents-gapless #91: Invisible click areas?

allstar87, 2018-10-07:
I'm building a touch screen radiobuttonshow, and I want to overlay invisible click areas over the screen.  I don't want borders or images associated with them; my background image will make it clear where users can click.
When starting the show, it throws an error saying "missing image fields in screen.cfg, see 1.3 release notes."  All I see in there is that click areas "can" have images.  Does that mean they must have images?

KenT2, 2018-10-08:
Can is the case. To get a blank image and avoid the error each click area must have the following in screen.cfg:

image =     
image-width = 
image-height =

I need to improve the manual which does not say this

KenT2, 2019-04-15:
I could not produce your error. Manual improved in 1.3.5d

closed 2019-04-15 by KenT2