KenT2/pipresents-gapless #92: Count number of Button clicks

HoneyBearUK, 2018-11-04:
Hi Ken,
I have 10 shows running on my setup each started by a button press and it would be good to know their popularity.
Is their a way I can count the number of times a button was pressed to start a show and record to a text file?
I do not need to know each request time, just a total for the day.
Many thanks,
Stewart

KenT2, 2018-11-04:
See Manual section 22.1  - Statistics Production. The other possibility is Counters

HoneyBearUK, 2018-11-04:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/KenT2/pipresents-gapless/issues/92#issuecomment-435702236",
> "url": "https://github.com/KenT2/pipresents-gapless/issues/92#issuecomment-435702236",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [KenT2/pipresents-gapless] Count number of Button clicks (#92)",
> "sections": [
> {
> "text": "",
> "activityTitle": "**Ken**",
> "activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
> "activitySubtitle": "@KenT2",
> "facts": [
>
> ]
> }
> ],
> {
> "name": "Add a comment",
> "@type": "ActionCard",
> "inputs": [
> {
> "isMultiLine": true,
> "@type": "TextInput",
> "id": "IssueComment",
> "isRequired": false
> }
> ],
> "actions": [
> {
> "name": "Comment",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"KenT2/pipresents-gapless\",\n\"issueId\": 92,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
> }
> ]
> },
> {
> "name": "Close issue",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"KenT2/pipresents-gapless\",\n\"issueId\": 92\n}"
> },
> {
> "targets": [
> {
> "os": "default",
> "uri": "https://github.com/KenT2/pipresents-gapless/issues/92#issuecomment-435702236"
> }
> ],
> "@type": "OpenUri",
> "name": "View on GitHub"
> },
> {
> "name": "Unsubscribe",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 406180686\n}"
> }
> ],
> "themeColor": "26292E"
> }
> ]

Hi Ken,I've downloaded the latest version now and that looks good.Many thanks.Sent from Samsung tablet.

HoneyBearUK, 2018-11-05:
Ken, many thanks the in build Statistics are great.

closed 2018-11-05 by HoneyBearUK