KenT2/pipresents-gapless #93: Using Radio Buttons but would like to return to different points in a Main loop

HoneyBearUK, 2018-11-06:
Hi Ken,
Hopefully you can give me some pointers for the below :- 

I'm using the current version of PiPresents  - 1.3.5c

My program has been working fine for some months. It consists of a RadioButtonShow (10 buttons) and a 'Main_Loop of some 200 + images.
Currently when returning from any of the '10' selections (Sel_1 to Sel_10) the 'Main_Loop' starts at slid one.

I would like to remodel my solution so when returning from 1 of the 10 selections i start at a different point in the 200 images.

My thoughts are:-

1. Split the 200 images into 10 sets of 20 images i.e. 'Main_Loop_1 to Main_loop_10' and at the end of a Sel_x the program would go to one of the 'Loops' and then continue to the next loop and the next loop (like the loop was never broken) until another Sel_x button is pressed.

2. Leave the 'Main_Loop' in tack and when returning from one of the Sel_x's go to a specified slide number. i.e. Come out of Sel_2 then open Main_Loop play Slide 40.

Any guidance would be much appreciated, I've tried many combinations but with no success.

Keep up the good work with the marvelous program.
Many thanks,
Stewart

KenT2, 2018-11-07:
Hi
I think this will work, you need the very latest version of gapless  which you have.
You need 10 radiobuttonshows, they are identical except that each has  one tenth of the  images in the main loop.
Each track played from the main show must be a show and 'Allow controls in subshows' needs to be enabled in the radiobuttonshow
In  the shows played from the radiobuttonshow use the 'show control on event' field to define pp-stop to 
be 'openexclusive show-x' where show-x is one of the other radiobuttonshows.

HoneyBearUK, 2018-11-07:
Hi Ken,
Thanks, I think i understand.
I will give it a try in the next few evenings and let you know how it goes.

HoneyBearUK, 2018-11-08:
Hi Ken,

 

I’ve been trying to get this going for a few hours now with little success.

 

To keep my testing simple to start with, I have 3 RadioButtonShows, Radio_Test_1, Radio_Test_2 and Radio_Test_3, each with 4 tacks as Shows in them. 

Track one is mapped to the a media file containing 1/10 of the images (well just four at the moment) named Loop_1, Loop_2 and Loop_3. 

The other 3 tracks in each RadioButtonShow are mapped to Show_1, Show_2 and Show_3. 

These 3 shows just contain some Message slides to tell me where I am. 

In each RadioButtonShow in ‘Controls’ I have the links for Show_1, Show_2 and Show_3.

 

At this stage I can test each RadioButtonShows as a separate unit and they play the Loop_x and will play the Show_x’s on demand.

 

If I now set ‘Show – Controls in Subshow’ to ‘Yes’ the Shows no longer start. The program I’ve been using for some time is also set on ‘No’ and the Show_x’s work fine, so I have left it set to ‘No’ for now.

 

Now the next bit I’m not convinced I’m doing what you are suggested.

 

radiobuttonshow use the 'show control on event' field to define pp-stop to be 'openexclusive show-x'

 

By ‘Show Control on Event’ do you mean ‘Show Control at beginning’ –

 

 

I have tried entering ‘openexclusive Radio_Test_x’ in the indicated box and the whole programme just starts and finishes in a spit second as it runs through the shows showing just a blank (black) screen.

 

If I place the above string in the ‘Show Control at End’ box the program will loop through each Loop_x as expected but the Shows no longer Start on demand I just get a frozen Black screen.

 

Have I just total misinterpreted the last part of your mail?

 

Hopefully the above will make some sense to you.

 

Many thanks.

Best regards,

Stewart

 

 

p.s. At least I understand the program and it working a lot better know, it’s surprising what six hours of trying things can do 😊

KenT2, 2018-11-10:
My mistake, you need Version 1.4.1 of PP (pipresents-beep) to get show control on event. I have prepared a little profile that should do what you want. Its different to my description in the previous  reply as I realised there is a simpler way to do it. 
[honey.zip](https://github.com/KenT2/pipresents-gapless/files/2568356/honey.zip)

KenT2, 2018-11-22:
Show Control on Event was added almost by mistake. It was a failed attempt to implement beeps but was too slow for that purpose. However now it's there I can see it's a potentially very powerful feature to make all sorts of exotic applications

closed 2018-11-22 by KenT2