KenT2/pipresents-gapless #97: Presentation question looping video to trigger one time video

Alchemist45, 2018-11-21:
Hi Ken,

I want to thank you again for a splendid program for small museums.  At Old Independence Regional Museum, a small regional non-profit museum, we are using PiPresents for several applications.  

I have a new request from our leader.  She wants a looping short video that can be interrupted by a GPIO button push to trigger a one shot longer video, which on ending flips back to the looping short video and waits to trigger again.

I cannot tell if this can be fitted to any of the existing examples and hoped you could point me in an appropriate direction.

Thanks in advance,
Scott

Alchemist45, 2018-11-21:
Hi Ken,

After looking around, I believe that the answers to issue #86 Attract/Main video may be my answer.  What do you think?

Thanks.
Scott

KenT2, 2018-11-21:
Yes, that was what I was going to suggest

Alchemist45, 2018-11-30:
Hi Ken,

I am still having trouble implementing the solution.  Can you contact  or put me in contact with the user lwshuck who apparently had success with the Attract/Main video solution?  If I could see a copy of the successful profile/profiles he used, it would help.  Obviously I don't need or want the private media he is using, only the profile structure.  If this is not available, can you please give me some additional pointers in setting this up with a repeating main show with a triggered one-shot show that returns to the repeating show?

Thanks in advance for any advice/help you may be able to offer.  Thanks again for a splendid program.  I guess I'm just not great at reading your documentation for this application.  My fault entirely.

Thanks again.

Alchemist45

Alchemist45, 2018-11-30:
Sorry, 
Accidentally closed  issue.

closed 2019-04-15 by KenT2