KenT2/pipresents-gapless #98: Feature request: Support InfluxDB for statistics

wdonker, 2018-12-09:
Hi,

In our museum we use Influxdb/Grafana to store and report on our indoor-climate measurements for some time now.
Now we like to use Influxdb also for storing statistics on usage of PiPresents. I already managed to adopt pp_utils.py to send a HTTP-post to Influxdb but it is rather hardcoded. and it would be nice if it is configurable through the web-editor.
Maybe this is a feature that also has value for other users of PiPresents.

KenT2, 2018-12-12:
Hi
Sounds like a good idea.
There could be many different ways to write out the data so  probably the best way is for me to separate out that part of the stats logging that does the writing out into a separate class/module which users can rewrite to their own requirement ad then insert into any update of PP.

Could you post your modified code here so I can see what the module's interface should be.

wdonker, 2018-12-13:
Hi Ken,

something like a plugin is fine I think.
Influxdb has an easy HTTP-API that can be used to http-post entries to the database: [//docs.influxdata.com/influxdb/v1.7/write_protocols/line_protocol_tutorial/](https://docs.influxdata.com/influxdb/v1.7/write_protocols/line_protocol_tutorial/)

I think it would be enough if we can specify the URL to that API and a message template that contains variables to be substituted by PiPresents, e.g.:
URL: http://localhost:8086'/write?db=pipresents
Message: <some_measurement>,profile=$profile$,show=$show$,track=$track$ value=1

I attached my adapted version of utils.py. Changes are marked with my initials (#wd) and are mostly in the stats-procedure (apart from a small change in start-stats because the profile name was not in the arglist for stats).

[pp_utils interfacing influxdb.py.txt](https://github.com/KenT2/pipresents-gapless/files/2675383/pp_utils.interfacing.influxdb.py.txt)

KenT2, 2019-04-27:
Implemented in Version 1.4.2a (pipresents-beep]. Untested module to interface with Influxdb attached. 

[pp_statsrecorder.py.txt](https://github.com/KenT2/pipresents-gapless/files/3124362/pp_statsrecorder.py.txt)

closed 2019-04-27 by KenT2