KenT2/pipresents-gapless #99: Multiple GPIO Input to start a show

KampeK, 2019-01-12:
Hi,
I would like to make a show where I have 9 buttons connected with GPIO. A spesific picture is shown if 3 buttons are pushed/aktivated at the same time. An other picture is shown when a different combination of buttons is aktivated.
Is there a posibility to do so? I can imagine that it needs some programming, how could I do that?
Thankfull for any suggestions.
Thanks,
Karen

KampeK, 2019-01-13:
Thank you for your answer. I am a beginner in programming and in pipresents and raspberry pi. I have never used an Arduiono before. 
I found the _do_buttons self_callback, but unfortunately I have no idea how to modify it in the right way. I want to activate three buttons at a time. When a new button is pressed a new action is triggered. Where exactly do I change the code and how? Could you give me an example? I would really appreciate it.
Thanks a lot!

KenT2, 2019-01-19:
I have given the problem some thought and its quite tricky, which is why people do not use this method of input. The problem comes because there will be times when transiently only 1 or 2 of the 3 required buttons are pressed so the software will need to detect that the transient condition is over in all possile situations. The easiest way to do this is to have a fourth 'GO' button which is  pressed when the selection has been made.

A better solution would be to use a numeric keyboard. There are probably ones out there that look like ordinary keyboards without letter keys. There may also be ones that produce ASCII codes but interface through USB. These would probably interface to PP using the pp_serialdriver.py

KampeK, 2019-01-21:
It is really kind that you are giving my problem some thoughts. 
I am working at a nature information center in Sweden and I am really impressed by your software. We are building a new part of an exhibition and I am trying to realize my ideas with pi presents. I think I have solved some of my ideas but some remain. And I really appreciate your help!
If I understand you right it would work that I press three keys on a keyboard at the same time to generate one event, and if I then press three other keys I could generate a new event? Where and how can I program this?
In this case I could use an old USB keyboard and rebuild it so that the keys are buttons instead. I don´t necessarily need to use the GPIO inputs, if it is too complicated!
I am rebuilding a keyboard for another project. In that project I want to have a big map of our region on the wall, with approximately 25 buttons integrated. Each press of a button generates a picture of the area on a screen. I thought of using a radiobuttonshow but instead of buttons connected to GPIO inputs I thought of rebuilding a USB keyboard and using them as inputs. 
For this purpose, I have been trying to find a possibility to define new keyboard actions to a radiobuttonshow but unfortunately, I failed. How do I do it? I tried to add the keys.cfg in the pp_io_config of a radiobuttonshow but the program is not reacting on the keys defined (e.g. m = pp-mute). I even tried to find the definition of but1, but2, but3 and but4 in radiobuttonshow in order to define more buttons but I can´t find the definition file.

KenT2, 2019-01-23:
>If I understand you right it would work that I press three keys on a keyboard at the same time to generate one event, and if I then press three other keys I could generate a new event? Where and how can I program this?

You misunderstand me, a button is a simple contact whereas a key on a keyboard generates a code from effectively 8 internal buttons in each key. What I suggested is pushing a single key on the keyboard for each button.

>In this case I could use an old USB keyboard and rebuild it so that the keys are buttons instead. I don´t necessarily need to use the GPIO inputs, if it is too complicated!

As above, keys on keyboards are not just buttons. My suggestion was to use a standard numeric keyboard which has an IC in it to decode the matrix output from a button into a key code. You would need to experiment with the numeric part of an ordinary keyboard to see if PP supports it.

> I am rebuilding a keyboard for another project. In that project I want to have a big map of our region on the wall, with approximately 25 buttons integrated. Each press of a button generates a [...]

Better to have 25 buttons which the Pi has.

> For this purpose, I have been trying to find a possibility to define new keyboard actions to a radiobuttonshow but unfortunately, I failed. How do I do it? I tried to add the keys.cfg in the [...]

In the Controls tab of the Radiobuttonshow the controls field should have but1 play track1

KampeK, 2019-02-04:
Thanks for the explanation of pp-key-* it works now, which is great!

I can´t really give up my other idea though… 
I have one more question, because I think that we may missunderstand each other.
I wrote a very easy litte program to test my idea. I got a LED to light by activating three magnetic switches one after one. I now wonder if I could translate this to pipresents. Instead of lightening a LED I would like pipresents to open a file.
Here is my litte program as an explanation:

from time import sleep
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
button1=12
button2=7
button3=16
button4=15
LED1=22
LED2=11

GPIO.setup(button1,GPIO.IN,pull_up_down=GPIO.PUD_UP)
GPIO.setup(button2,GPIO.IN,pull_up_down=GPIO.PUD_UP)
GPIO.setup(button3,GPIO.IN,pull_up_down=GPIO.PUD_UP)
GPIO.setup(button4,GPIO.IN,pull_up_down=GPIO.PUD_UP)
GPIO.setup(LED1,GPIO.OUT)
GPIO.setup(LED2,GPIO.OUT)

BS1=False
BS2=False

try:
   	while(1):
		if (GPIO.input(button1)==0 and GPIO.input(button2)==0 and GPIO.input(button3)==0):
			print("Button 1 was Pressed")
            		if BS1==False:
                	GPIO.output(LED1,True)
             		BS1=True
                        sleep(0.5)
       	else:
            	GPIO.output(LED1,False)
       		BS1=False
            	sleep(0.5)
            
        if (GPIO.input(button2)==0 and GPIO.input(button1)==0 and GPIO.input(button4)==0):
            	print("Button 2 was Pressed")
         	if BS2==False:
                GPIO.output(LED2,True)
                BS2=True
                sleep(0.5)

	else:
            GPIO.output(LED2,False)
            BS2=False
           sleep(0.5)
finally:
    GPIO.cleanup()

KenT2, 2019-02-04:
Now add
    if (GPIO.input(button2)==0 and GPIO.input(button1)==0 :
        	print("Code 3 was produced")
     	if BS3==False:
            GPIO.output(LED3,True)
            BS3=True
            sleep(0.5)

and note that you are trying to produce 8 codes from three buttons
 You may find that unless you press two buttons at exactly the same time you produce code 1 or 2 instead of 3 and will have erroneously started video 1 or 2 instead of 3. You may be saved by the half second delay which disguises the problem. 

The code you have written is the approach I suggested initially to add what you want to pp_gpiodriver.py In _do_buttons pin[pp_gpiodriver.PRESSED] is the de-bounced state of the button and replace the print statements with a self.button_callback function call

closed 2019-03-20 by KenT2