KenT2/pipresents-gapless #101: Show control 'exitpipresents' at end of topshow causes loop

wdonker, 2019-02-09:
When I add the command 'exitpipresents' in 'show control at end' of the topshow it does not end PiPresents but causes a loop and eventually crash because of maximum recursion depth.
It seems PiPresents tries to end all shows over and over again, triggered from the topshow.

Extract from the debeug log (topshow had a duration of 30 minutes):

1801.88 VideoPlayer:       Entering state : closed from show Id: 3
1801.89 MediaShow: Closed after showing track with reason: normal and message: omxplayer closed
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received
1801.90 PiPresents: command received: exitpipresents
1801.90 ShowManager: De-registering show images-huisjes show index:0
1801.90 ShowManager: De-registering show images-timeout show index:1
1801.90 ShowManager: De-registering show mainshow show index:2
1801.90 ShowManager: De-registering show timeout show index:3
1801.90 ShowManager: Exiting show timeout show index:3
1801.90 MediaShow: timeout 3: Exit received

wdonker, 2019-02-09:
I may have found a solution already but I'm not sure.
I changed pp_showmanager.py and added the following after line 165:
`            self.set_exited(index)`

This ended the loop.

KenT2, 2019-02-16:
Looks like this will work in most cases but maybe not all. PP should wait until the show has closed all its tracks and subshows before doing set_exited(). I'm not sure whether this is missing call or whether having an exitpresents in show control at end is causing a circularity.
It would be useful if you could post you profile here (without media) or contact me via wordpress so we can talk by email.

wdonker, 2019-03-05:
Hi Ken,

here's the profile concerned:
[hm_koren_op_de_molen_segment_3.zip](https://github.com/KenT2/pipresents-gapless/files/2930393/hm_koren_op_de_molen_segment_3.zip)

KenT2, 2019-04-15:
fixed in 1.3.5d. The cause was the circularity in having exitpipresents in the show control for a show

closed 2019-04-15 by KenT2