KenT2/pipresents-gapless #102: Pressing Exit in pp_manager throws TypeError

gzotti, 2019-02-11:
Hi! First let me thank you for this excellent system. 

I am developing a setup with 1.3.5c, with two RPi Zero W connected and synchronized via OSC in their own WiFi. All works so far, and they are simply powered by the TV USB plugs. To shut down cleanly or for simple maintenance operations I'd like to let staff use a tablet pointing at the respective pp_manager web GUIs. However, if I want to stop a running mediashow (say, pp_mediashow_1p3) pressing Exit (bottom/central), the console running pp_manager shows:

     Traceback (most recent calls last):
       File "/home/pi/pipresents/pipresents.py", line 882, in <module>
         pp = PiPresents()
       File "/home/pi/pipresents/pipresents.py", line 447, in __init__
         self.root.mainloop( )
       File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1133, in mainloop
         self.tk.mainloop(n)
      File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1540, in __call__
        def __call__(self, *args):
     TypeError: handle_sigterm() takes exactly 2 arguments (3 given)

and obviously the system is then locked in a state of "pipresents is running". Shutdown and reboot is still working, but it is not possible to exit a running mediashow via pp_manager's web GUI.

KenT2, 2019-02-11:
handle_sigterm is missing an argument. Change line 651 of pipresents.py from 
def handle_sigterm(self,signum):
to  
def handle_sigterm(self,signum,frame):

gzotti, 2019-02-12:
Yes, that worked! Thank you again!

KenT2, 2019-04-15:
fixed in 1.3.5d

closed 2019-04-15 by KenT2