KenT2/pipresents-gapless #106: Feature request (?) play/stop etc controls by MIDI

garubi, 2019-03-05:
Waiting for my Raspberry PI to arrives, I'm studying the documenattion... 
But I can't understand if and how I can config Pi Presenters to be controlled by sendig to it MIDI commands (Note on/oof or COntrol Changes etc)

We are aiming at using Pi Presenters during our band's shows and would be nice to start/stop background videos play and images display from a spare MIDI controller.

If MIDI controls is not implemented is there a way to integrate it (maybe a plugin or a future relesease)?

Thank you 

Stefano

KenT2, 2019-03-07:
PP does not support midi input however it does support OSC. You could either find  a midi to OSC convertor or modify pp_oscdriver to support midi instead of OSC.

garubi, 2019-03-08:
Thank you Ken!
I'm waiting the arrives of a new Pi (since looks like I can't run PP on a pi desktop distribution, can you confirm this?) And try one of the two solutions

KenT2, 2019-03-08:
The core of PPincluding displaying images might work but videos and GPIO probably will not . Audio might.
I've not tried it. It would be interesting to do the experiment while you are waiting.

garubi, 2019-03-08:
OK, I'll do...
first of all I don't think that omxplayer could be built on intel chipset
... :-(

KenT2, 2019-03-08:
But mplayer2 or mpv might be able to run on intel H/W. Probalby best way to integrate it is to modify pp_omxdriver.

garubi, 2019-03-11:
thank you @KenT2 
maybe I'll give it a try... or just wait some days and directly go with the incoming Rasperry

closed 2019-08-07 by KenT2