KenT2/pipresents-gapless #113: Problem with pp_io_config

Abanoubel, 2019-07-09:
I am trying to have a simple presentation whenever I press a button it shows a certain video and whenever I press the other button it plays another video. these buttons are connected to the pins. The problem is with any profile that has pp_io_config I get an error, if it it is my profile or the examples profile I get the same error

Here what happens when I run pp_animate_1p3
pi@raspberrypi ~ $ python /home/pi/pipresents/pipresents.py -p pp_animate_1p3 -f -b
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 882, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 370, in __init__
    reason,message=self.ioplugin_manager.init(self.pp_dir,self.pp_profile,self.root,self.handle_input_event)
  File "/home/pi/pipresents/pp_iopluginmanager.py", line 26, in init
    status,message=self.init_config(cfgfile,cfgfilepath,widget,callback)
  File "/home/pi/pipresents/pp_iopluginmanager.py", line 63, in init_config
    reason,message=instance.init(cfgfile,cfgfilepath,widget,callback)
  File "/home/pi/pipresents/pp_io_plugins/pp_gpiodriver.py", line 172, in init
    self.GPIO.setup(num,self.GPIO.IN,pull_up_down=pin[pp_gpiodriver.PULL])
RuntimeError: No access to /dev/mem.  Try running as root!

IF I run it as root I get that
sudo python /home/pi/pipresents/pipresents.py -p pp_animate_1p3
Do not run Pi Presents with sudo

KenT2, 2019-07-09:
This should not happen. sudo was not required since a few years ago.
Which version of Raspbian are you using?
Is it a clean install of Raspbian or have you changed any permissions etc.

Abanoubel, 2019-07-09:
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 9.9 (n/a)
Release: 9.9
Codename: n/a

KenT2, 2019-07-09:
I don't recognise this version of Raspbian.
You need the Raspberry Pi Foundation version [https://www.raspberrypi.org/downloads/raspbian/](url)

Raspbian: Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage4

I am still usig a Stretch based version, the latest Bster version should work OK on a Pi3

I suspect you have found the Raspbian version. Confusing isn't it!

closed 2019-08-07 by KenT2