KenT2/pipresents-gapless #114: PPManager through remote access does not load

boschwakka, 2019-07-23:
Hi there
Loaded pipresents on a Pi 3 B+ working great have got the pi talking to wifi static ip and links to internet. Setup dsl router for remote access with dyndns name and opened ports 22 for ssh and 8081 and 8082 for web access, on local network web access works well, opens in firefox and can stop, start and run editor ok. BUT through remote access i just get the folding square running. Remote ssh works fine connecting to port 22
[image]

KenT2, 2019-07-23:
Can't help with this. However it looks from the terminal window running the manager that it uses additional random ports for websockets.

Pi Presents uses Remi [https://github.com/dddomodossola/remi](url) you could ask there for advice. Davide's advice in his readme is to be careful if exposing a remi application to the internet

boschwakka, 2019-07-24:
Thanks very much KenT2 will update the post with what i can find out from remi.

boschwakka, 2019-07-25:
Hi kenT2 got a reply back from Davide and he suggests updating remi to the latest version as there are fixes from the version you are using. What do i need to do to upgrade his version into your pi-presents.
can you assist me please. thanks mark

KenT2, 2019-07-25:
The directory /pipresents/remi  contains all the remi code. Just replace the contents of this with the code from github keeping the same revised structure.

The version I encapsulated in pipresents is quite old so the API might have changed. I didn't intend it to be updated by users but give it a try

KenT2, 2021-07-04:
In **pipresents-beep** I  updated Remi to its latest version on 2/4/2020 so it should have Davide's fixes.

closed 2021-07-04 by KenT2