KenT2/pipresents-gapless #115: Autostart Pi Presents instructions not working

mercuryrsng, 2019-08-07:
Hello,

I am a new Raspberry Pi/Linux user and I am trying to use the instructions in your guide on how to autostart Pi Presents.  I am currently running Raspbian Buster.
In the instructions, it states that the autostart file is located:
/home/pi/.config/lxsession/LXDE-pi/autostart

The problem is the lxsession folder does not exist.  I have to imagine that in later versions, they have changed the location of the autostart file.

Where can I find it now?  Or, how can I get my profile to start up on boot?
Please go easy on me as I am still learning bash :-).

Love the product!

mercuryrsng, 2019-08-07:
NEVERMIND all I had to do was slow down and read.....
Works like a CHARM....thanks!!!

KenT2, 2019-08-07:
autostarting pp_manager.py uses the same technique as starting PP

mercuryrsng, 2019-08-07:
I figured it out. Thanks again!

closed 2019-08-07 by KenT2