KenT2/pipresents-gapless #116: another question about: Attract/Main video Museum Exhibit

tejo23, 2019-08-28:
Hi Ken,

Thanks for creating Pipresents. 

I'm a Dutch artist creating installations. I wanted to do the same as "lwshuck". Create a loop to attrack visitors and when they push a button a video starts.

I've got it working in one mediashow with a child track and at the controls PIR play. 

This will trigger the main video. Which is good. 

The issue is:

When I push the button while the video is playing, it restarts. I want the button to be inactive for as long as the main video plays. As soon as the video has ended and the loop is playing again, people can start the mail video again.

Is it possible with pipresent to not restart the main video (which is in the child track)?

Thank you so much for your answer.

Kind regards 
Frank Rigters
The Netherlands

KenT2, 2019-08-28:
In the main track put the control play_main null. Where play_main is the symbolic name you used to start the video.

The control in the show is then overriden with a 'do nothing' while the video is playing

tejo23, 2019-08-28:
Hi Ken,

Thank you for your quick response. I tried what you suggested but I don't get it to work...

My setup:

Shows: 
- Start [start]
- video [video]
       *child track field : main
       *controls: PIR play

medialist:
- media.json

Tracks:
- loop
- main [main]

I don't get to the right field to implementate de play_main null as you suggested. I tried in the controls-tab in the editshow, and in de controls-tab in in the edit of the track "main". 

It both didn't work.

sorry for my lack of programm knowledge... 

Thanks 
Frank

KenT2, 2019-08-29:
you need

PIR null in the controls field of the main track

tejo23, 2019-08-29:
Dear Ken,

Thank you so much for your support! It works now exactly like I want it to!

Kind regards
Frank

closed 2021-07-04 by KenT2