KenT2/pipresents-gapless #117: ImageTK

pramodmadhav, 2019-09-23:
after running pipresents.py i get following error 
ImportError : cannot import name imagetk
I am using raspberry pi 4 with raspbian buster

KenT2, 2019-09-23:
Sounds like you have missed out an installation step:
     sudo apt-get install python-imaging-tk

pramodmadhav, 2019-09-24:
Ok let me do the installation process once again and will let you know.
But as far as my knowledge I have done that
Is it something to do with is (buster)?

pramodmadhav, 2019-09-24:
Problem solved
Instead of python-imaging-tk I had to type
Python-pil.imagtk
May be because I installed python-pil instead of python-imaging which was prompted at the time of installation
Is it ok or not I don't know

pramodmadhav, 2019-09-24:
now the error window comes with profile not specified as explained in installation procedure.
now when try to down load the examples with"         wget https://github.com/KenT2/pipresents-gapless-examples/tarball/master -O - | tar xz"
I get following error message
HTTP request sent awaiting response  404 not found
what could be the reason?

pramodmadhav, 2019-10-05:
I could down load the examples and was able to see the examples.
Also read your manual.I saw ocs in that. 
can PiPresents sync multiple videos on multiple raspberries?may be using OSC

KenT2, 2019-10-06:
Yes OSC will do it however it depends on the accuracy of the sync you require. There will a delay of a second, probably less. You might be able to use the video playout to improve it.

For 2 videos you might be able to  use a PI4 with 2 monitors, arrange them one on top of the other using the screen configuration utility.

Something like Pi-Wall will do a frame sync IIRC

pramodmadhav, 2019-10-20:
thanks,
I came accross follwing link to sync the videos on multiple rpis,
it works well for me on 9 nos of rpi without any problem
if I want to integarte that in pi presents can you tell me how to do it.
https://github.com/turingmachine/omxplayer-sync

KenT2, 2019-10-20:
Looks an interesting package. 
omxdriver is called in pp_omxdriver.py. It then uses dbus to pause the video and to monitor the position of the video to freeze it at the end; both part of achieveing gapless operation.

It might be better to replace the whole of pp_omxdriver.py with software based on omxplayer-sync ideas.

pramodmadhav, 2019-10-25:
What about pp-omxdriver.pyc?.
frankly speaking I do not know all these things.
Only if you could add this as package or additional feature to pi presents that will be very useful.

KenT2, 2019-10-27:
Transferred to Beep

closed 2019-10-27 by KenT2