KenT2/pipresents-gapless #120: How to advance tracks by keypress and/or play one with specific key

garubi, 2019-11-20:
Hallo Ken,
first of all thank you for this great software!

We are two musicians who performs with an actress, our show are a mix of music and historical tales about great women of '900.
We have used Pipresents with success this summer to display pictures and videos during the shows.
You can see pipresents in action in the show about Nobel's prize [Margherita Hack](http://margheritatralestelle.it/) and in the show on [Alfonsina](http://alfonsinacorridora.it/), the first women who participate in the Giro d'Italia race.

**I can't find a way to do the following, do you have some suggestion?**
- We know that using the **mediashow** we can have a "ordered" list of tracks that start one after another with the press of one key. When a track ends, Pipresents wait for the key press then starts the next track. 
_This mode is perfect when we have a screenplay to follow (and he actress doesn't forget it ;-) )_ 
- On the other hand, we also know that using **radiobuttonshow** we can assign a specific key to each track, too and -if needed- start that track at the press of that specific key (let's say: start track 1 with key "a", start track 5 with key "f", and so on...). 
_We use this mode when we play a gig and we are not sure of the song's order: when the bandleader call a song, I press the key specific to that background video of that song._

**The question:
Is there a way to combine this two "style of control"?**
i.e. : in normal situation we just advance from a track to another with the "next" key/command, but if needed have each track binded to its own specific key to start it "out of the standard order"?

Sorry for my English, I'll be more than happy to provide more informations if needed.

Thanks in advance

Stefano

KenT2, 2019-11-20:
No out of the box soultion to your requirement, however there are two possibilities.

One solution would be to use a menushow. To play the next track would need two commands, down and play. However if using GPIO the down could be on the leading edge of the press and the play on the trailing edge. (Could be a problem with reliability). For rehearsal just keep down and play as seperate buttons/keys.

The other solution is to use a mediahow for the show and a menushow for rehearsals. As all tracks are anonymous both shows can use the same medialist to save duplicating that (hopefully no funny side effects as I have not tried this since the original PP six years ago).

garubi, 2019-11-20:
Ken,
thank you for your answer.
I'll give it some thoughts and let you know how we solved

STefano

closed 2021-07-04 by KenT2