KenT2/pipresents-gapless #121: Pipresents shows profile error!

shobi2224, 2020-01-20:
Hi Ken,
I'm facing an issue with the pi-presents,
The issue is,
I'm just running the live tracks and it is working fine but whenever I'm deleting(delete the old files and update the display with the new files) the files in the live tracks folder while the display is running. It shows a profile error.
For example: In the live tracks folder there are 5 images,
1. rose.jpg
2. cars.jpg
3. fruits.jpg
4. mountain.jpg
5. colour.jpg
The display is running with the above images. 
When I'm deleting an image called fruits.jpg and at that time the display is yet to show the image but the pi presents isn't able to find the image as it is deleted, it shows the profile error. 
[image]

Kindly let me know how to resolve this issue.

Regards
Shobika R.

KenT2, 2020-01-20:
it is hotel 157-......transylvania (7).jpg that is causing the problem.
The content of the file may be corrupted, try it as one of the initial files in live_tracks

PP may not like spaces in file namesrmove them

Try renaming 157_transilvania so it does not appear first in the list.

shobi2224, 2020-01-20:
So you mean to say that it will not create a profile error if I delete a running file but it will show an error if the file name is improper. Am I right?

KenT2, 2020-01-20:
yes

shobi2224, 2020-01-20:
Ok thanks, let me try and get back to you tomorrow!

shobi2224, 2020-01-22:
Hi,
I have tried with 2 scenarios, 
1. I'm having a cloud which pushes the files(Images/Videos) to the Raspberry pi in a temporary folder. 
And in Raspberry pi, I've written a code to push the files from the temporary folder to the live tracks folder. In this scenario, the profile error is occurring sometimes.

2. When I'm deleting and adding the files internally then the pi presents is working fine. But while pushing the files from cloud then it shows the profile error sometimes. 

What might be the issue? Does the pi presents is compatible to work with the cloud? Or Should I need to do any configurations with the pi presents?

KenT2, 2020-01-22:
Scenario 1 can produce errors because of the asynchronous working. The push changes the content/number of files while PP is reading them. The problem is worsened if you are re-using filenames.

I think the answer is to copy the files from the temporary directory to the live tracks directory after line 251 of pp_livelist.py.

Of course this just moves the problem to you copying from the cloud to the temporary directory. I guess some sort of lock would be needed

shobi2224, 2020-01-24:
Thanks for your suggestion.
One doubt,
You want me to place the code(which I've written) after line 251 of pp_livelist.py. Am I right?

shobi2224, 2020-01-24:
Thanks for your suggestion.

Below is my code to move the files from the temporary folder to live tracks

import glob, os, shutil, time
while True:
    try:
        last = glob.glob('/home/pi/Temp_Files/*')
        x = len(last)
        time.sleep(30)
        current = glob.glob('/home/pi/Temp_Files/*')
        y = len(current)
        if (x != y):
            path = glob.glob('/home/pi/pp_home/pp_live_tracks/*')
            for f in path:
                os.remove(f)
            time.sleep(1)
          
            source_dir = '/home/pi/Temp_Files/' #Path where your files are at the moment
            dst = '/home/pi/pp_home/pp_live_tracks/' #Path you want to move your files to
            files = os.listdir(source_dir)
            for f in files:
                shutil.move(source_dir+f, dst)
            print "moved!"
        else:
             print "No change!"           
    except KeyboardInterrupt:
        GPIO.cleanup()

Now my question is, you want me to place the above code after line 251 of pp_livelist.py ?
Will this solve my issue?

shobi2224, 2020-01-25:
Hi, Is there any possibility that the video files playing will flicker(playing a single video, one above the one). So it seems like the video is flickering. I've set the resolution as 1920x1080

KenT2, 2020-01-25:
I need to look at the code to improve intgration with PP but yes, it goes after line 251.
If you play two videos at the same time one on top of the other they will flicker. This can be stopped if they are displayed on different layers, use the --layer option in the video track. See omxplayer github for details

shobi2224, 2020-01-27:
Hi, 
Thanks for your reply. 
But I'm not playing two videos at the same time. A single video is playing while at the same time the same video is automatically playing behind, which looks like flickering

For Example: 
In live tracks folder, I'm having 3 videos,
1. video1.mp4
2. video2.mp4
3. video3.mp4

Now when pp is running and video1.mp4 is playing, the same video1.mp4 is also playing behind the original video. So this looks like flickering. This is happening sometimes and not all the time.

Sometimes apart from the above flickering error also I'm facing the below error,
Profile error: omxplayer ended before loading track

KenT2, 2020-01-27:
Is this your additional code?

shobi2224, 2020-01-28:
Yes, the same code I've mentioned above (4days ago)

KenT2, 2021-07-04:
pipresents-beep now has an inbuilt locking mechanism for Synchronisation of Uploads

closed 2021-07-04 by KenT2