KenT2/pipresents-gapless #123: Looping video w/gpio trigger

ace2go, 2020-02-25:
I followed the tutorial pdf as follows : 

https://github.com/KenT2/pipresents-gapless-examples/blob/master/tutorials/gr_tutorial.pdf

All I get is a blank screen with "Press 1 for Track 1", and "Press 2 for Track 2" with "Press Escape to return here"
I added an edited gpio.cfg file into the profile
I am looking to complete what the tutorial suggests : an ongoing fullscreen screensaver, that upon gpio trigger will play video 1, video 2, video 3, and upon completion of playing that video, go back to the screensaver.

Is the GRKiosk1 file available somewhere so I can download and open to see what I am doing wrong?  Thanks a lot!

KenT2, 2020-02-27:
Instead of the GRKiosk1 use the pp_radiobuttonshow_1p3 profile in the examples. It sounds very similar
I noticed one mistake in the tutorial - last line.  Pin 13 play vid1 should be pin13 play vid1  , and for other pins.

ace2go, 2020-02-27:
I got it mostly working with the radiobuttonshow_1p3, thanks for the start!  I have a continually cycling working video working, and if I trigger but4, it plays another video, then returns to the main video, which is excellent.  How do I add additional videos?  I want to trigger, say "but3", so I clicked on add, added the video, altered the controls to pp-key-3-null and but3 null, as well as altered the show controls to have pp-key-3-play video and but3 play video, but it only plays the original video, not my added video...

KenT2, 2020-02-28:
using the pipresents editor check all the controls in the show and the track references in all the tracks
for x of 1 2 3 4
gpio.cfg has symbolic names of butx for the appropriate button
controls in show: butx play videox  
track references: videox 
control in track: butx null

closed 2021-07-04 by KenT2