KenT2/pipresents-gapless #124: High-pitched whine

Boombostak, 2020-04-08:
Hi there, I'm a new PiPresents-gapless user and I'm having a tricky problem.

I am running a Pi 4b 1GHz. Everything was working fine over HDMI. When using the 3.5mm to composite I get a steady tone at about 3.1kHz. This tone is in addition to any other audio that is playing back. If the image on screen changes, it introduces clicks that sound like a square wave. If the image is static it sounds like a 3.1kHz sine.

This only occurs when I launch PiPresents. It is not present at boot. It is not present if I play the videos using VLC or omxplayer. The tone persists after I exit PiPresents and it doesn't go away until I reboot. This noise occurs on any of the example presentations.

I have a presentation that uses a child show to switch between an attract video and a playback video using a pushbutton on GPIO pins 9 and 10. I am also using pins 5 and 6 for a shutdown/startup button. I have tried disconnecting these 

I have tried multiple power supplies and a battery. This has no effect on the noise. Moving and swapping out the 3.5mm to RCA cable has no effect.

Any advice or wild guessing is appreciated. Thanks.

KenT2, 2020-04-09:
Took me a while to work this one out but 625*50Hz is 3Khz (or 60 if in USA)

I suspect what you are hearing is the video coming from the 4 pin 2.5mm connector,
The connector suppiles video and two audio channels. Its pinout is a bit weird as ground is not the ring nearest the housing. Your ground connection for the audio might possibly be the video output.

[https://www.raspberrypi-spy.co.uk/2014/07/raspberry-pi-model-b-3-5mm-audiovideo-jack/](url)

I will investigate further why pp-gapless is turning on the A/V output. pipresents-beep has much better support for the Pi4.

Boombostak, 2020-04-09:
I checked my cable continuity and the TRRS is left, right, ground, video so it is wired correctly for the Pi. I suppose it's possible there is a short in the jack. However, since this issue only happens with PiPresents and not with VLC or omxplayer, I don't suspect a cabling issue.

It's worth noting that the whine is present on both right and left channels. My content is mono so I can't easily phase cancel it.

KenT2, 2020-04-09:
I tried pipresents gapless on a Pi4 with Buster just to check but as expected all was OK with pp_audio1p3

I have managed to generate a 3.1Khz signal on the TRRRS audio but only by enabling the composite video by adding enable_tvout=1 to config.txt and then running mplayer with an audio track started from an old CRT display. (Also got 3.1Khz from PP when  it runs an audio track which uses mplayer in this configuration)
Couple of possible causes:
Are you running Raspbian Buster, you must have this for Pi4
Have you added enable_tvout=1, unliekly as this turns off both the HDMI monitors

Plugging the video output from the TRRRS into an audio amplifier does not produce a 3.1Khz signal which disproves my wiring theory agreeing with you. Its more lke a 50Hz burble.

Boombostak, 2020-04-09:
I am running Buster.

I have been using enable_tvout=1 because this is for an exhibit using a composite monitor. My understanding is that on the Pi 4b you need to use enable_tvout=1 in order to use the composite output.

Everything runs fine over HDMI but when using composite I have this audio problem. Frustrating because it appeared so late in the development process.

I suppose as a workaround I can try a cheap HDMI to composite adapter.

KenT2, 2020-04-09:
Ah I should have read your initial email more carefully. You are hoping to use composite.
With composite enabled I can reproduce the problem just by running the command mplayer 01.mp3 from the media directory. So it's a mplayer problem. I will investigate further. If you find anything on Google that would be useful.
Few more avenues to think about as a wokaround , a USB to audio convertor, or a Pi3.

KenT2, 2020-04-10:
Think I have found the solution
Change your profile using the PI Presents editor
 - For audio tracks you need to change the audio output from hdmi to local
 - You also need to do the same for video tracks, otherwide the video seems to play very slow through omxplayer.
If that does not work disconnect the signal cable to all the hdmi monitors. It may be enough to power down the monitors.

I have tried PP-beep with the composite output. It crashes but it should be an easy fix which I will do soon- 

- 

- 

- 

-

Boombostak, 2020-04-14:
Success! Changing the 'video player audio' to local for all of the videos solves the issue and my presentation is now behaving as expected.

Thanks for the support!

KenT2, 2021-07-04:
Should also be fixed in Raspberry Pi Firmware
https://github.com/raspberrypi/firmware/issues/1373

closed 2021-07-04 by KenT2