KenT2/pipresents-gapless #127: ImageTK

s-m-arty, 2020-04-17:
after running pipresents.py i get following error
ImportError : cannot import name imagetk

KenT2, 2020-04-18:
you have missed out during installation

     sudo apt-get install python-imaging-tk

s-m-arty, 2020-04-18:
I have not forgotten this step. However, the following message appears in this step
`Fehl:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf python-pil.imagetk armhf 5.4.1-2
  404  Not Found [IP: 93.93.128.193 80]
E: Fehlschlag beim Holen von http://raspbian.raspberrypi.org/raspbian/pool/main/p/pillow/python-pil.imagetk_5.4.1-2_armhf.deb 404  Not Found [IP: 93.93.128.193 80]`

KenT2, 2020-04-18:
I installed this a couple fo days ago so I know it works, however it does not now so I suspect an intermittent problem with the server. Someone else reported this many months ago, try again later

s-m-arty, 2020-04-19:
after a new installation it works

closed 2020-04-19 by s-m-arty