KenT2/pipresents-gapless #128: Trouble with the example pp_magicpicture_1p3

Knightrider6224, 2020-04-29:
Hello !
First of all thank you so much for the incredible job you've made with PiPresents.
I was looking for Something like this for quite some times !!! Just brilliant.

I've tried almost every examples to familiarize myself with PiPresent but pp_magicpicture_1p3 does exactly what I want.
I have 6 videos starting and ending the same way, I just want them to run in a random way when a movement is detected by the PIR.
Everything is working fine but sometimes at the end of a video file the picture isn't freezing and I can see the background with my mouse pointer ( no pictures displayed ), it happens randomly and not always with the same file.
If the backgroud selected is black then I will see a black screen, I should normaly have the first frame of my video ( who is the same on all my videos ).
The program continues to run, I don't have a crash and if I trigger the PIR again a new video will be played.
Since my first frame on all those videos is the same, I've tried to export it as a JPG and put it as a background picture to get around this issue but the program doesn't display the JPG instead of the black backgroud.
If I change the backgroud color to something else than black, I will have the same issue but with the new color selected in PPWebEditor.
It's been two days that I'm trying to figure this out with no luck, I checked the files manually and I don't see why this is happening.
Any hints will be greatly appreciated.

Thanks

KenT2, 2020-04-30:
I suspect the pause at start (after first frame) is happening too early.

 Could you edit pp_omxdriver.py line 149  - after_first_frame_position to make the pause happen later. Its a negative number so make it closer to zero or positive if necessary. Remembering that at 25Hz each frame takes 40000 microseconds

Current text:
    after_first_frame_position = -50000 #microseconds
    before_first_frame_position = -80000 #microseconds

Keep a backup copy of the original file first

Other things for you to check:
Have you installed unclutter. Even with it installed the pointer appears for a while if you move the mouse
Are you using the -f command line option to give you full screen.
In the profile video track or using the track default in the show profile, is freeze at end  - yes and freeze at start - after first frame

Bit suprised the background image does not show. I notice that the background image can be in the show and in the track with the option in the track to disable the show background image. I might be worth experimenting with both of these options.

Knightrider6224, 2020-04-30:
Thanks for your answer.
I think you are correct in your assumption to modify pp_omxdriver.py, I've change the default values to those ones ::

after_first_frame_position = 30000 #microseconds
before_first_frame_position = 10000 #microseconds

I've triggered the PIR more than a hundred times now and so far I haven't seen anything else than what it supposed to be shown. 👍 

I don't know if this is related to the files I'm using but I'll keep in mind to modify those settings in the future.

Thank you very much for your support !

KenT2, 2020-04-30:
Its probably influenced by framerate, the power of the Pi, whether more than one show is running, and whatever other tasks the PI is doing. Its one of the few placed where I have to rely on a timing because the interface to omxplayer does not have a proper handshake.

I thought I had reviewed in for pipresents-beep but the code shows I didn't. 

Keep this issue open to remind me to review it in the next issue.

Knightrider6224, 2020-04-30:
I'll do that ! 

Thanks for the explanation about omxplayer

KenT2, 2021-07-03:
I am closing this because omxplayer is being replaced by VLC. pipresents-beep has VLC as its out of the box video player.

closed 2021-07-03 by KenT2