KenT2/pipresents-gapless #129: LED PWM fades

Boombostak, 2020-05-08:
I have a project that requires an LED that slowly fades up and down in intensity.

I'm not seeing a way to achieve this within PiPresents.

I have a pyton program that achieves this, but when PiPresents launches concurrently, the LED stops fading (usually in some intermediate level of brightness). I need to do more reading to better speculate what might be happening.

Just wondering if this is a problem that has been tackled in the past - a search of the issues did not turn up anything about PWM.

Thanks!

KenT2, 2020-05-08:
PP cannot fade the GPIO, although it could be added to pp_gpiodriver.py

Your fade probably stops because Pi Presents is also using the gpio. If you do not need to use the gpio in Pi Presents then disable it by removing gpio.cfg from /pp_io_config in the profile and /pp_io_config in /pipresents
If that does not solve the problem then tell me as it might be a bug.

Boombostak, 2020-05-08:
I do need to use GPIO to run a pushbutton. It's simple setup, one pushbutton that fires the video and one LED that fades in and out as an attract feature.

I'm wondering if there is a way to release specific pins from PiPresents? I don't really know the language for this as this is a new problem for me.

KenT2, 2020-05-08:
Pi Presents uses RPi.GPIO

if you set the direction of the pin to none in gpio.cfg PP does nothing with the pin.

PP does set the mode to Board, I don't know what happens if your program uses the other mode.

lalo-uy, 2020-05-13:
Maybe a Animation plugin that launches a separated program in background.
That way the fading program can be started, or any other one by the way.

KenT2, 2020-05-14:
I think I shall implement blink and fade by a pwm extension to pp_gpiodriver.py. However I like your idea of an I/O plugin that can run an external process. Might also be worth doing a track plugin that does the same.