KenT2/pipresents-gapless #130: Input-pending trigger on audio tracks

lalo-uy, 2020-05-13:
I whish to have the audio track preloaded and paused, since ther is a 3sec delay in playing.
Will work also in repeat?
The short audio runs in paralel with a presentation triggered with same event.

Many thanks in advance.

KenT2, 2020-05-14:
Video tracks have a Freeze at start and Go command which do what you want. However I cannot implement this for audio tracks due to limitations in mplayer.

 I think you can play an audio file in a video track and get the Freeze at start but I am not sure.
Another possibility is to use [Presents Beep] and the beep feature. This plays audio files using aplay which loads much quicker.

lalo-uy, 2020-05-15:
Present-beep do the trick.
Many thanks.

By the way, I extended counters functionality with load & save actions. The file lives in pp_home and only contains the number.
I use load at start of show and save on end comand.

closed 2020-05-15 by lalo-uy