KenT2/pipresents-gapless #131: using vlc player instead of omxplayer?

najoshi, 2020-06-28:
Hi Ken,

I want to use your software to create a digital media display for my house using the pi 4. I have done this before with pipresents-next. The display will show both pictures and video and I want to annotate both of them with text. This is easy with the pictures, but I am running into a problem with the videos. I used subtitles for the videos and apparently the raspberry pi 4 does not support the thing which omxplayer uses for subtitles. Also, omxplayer is being deprecated in lieu of vlc. So, my question is, is there a way to use vlc instead of omxplayer with pipresents-gapless? Are you planning on using vlc for pipresents-beep? Any help you can give me would be highly appreciated.

- Nik.

KenT2, 2020-06-29:
I am investigating adding VLC to Beep. At the moment VLC is unuseable for this as the 'fullscreen' option is always fullscreen and obliterates any images and text behind it. Pi Towers is currently working on this.

I do have a proof of concept python program which is working well other than the above so I am hopeful that this might happen sometime.

najoshi, 2020-06-29:
Well according to the omxplayer github page:

> Note: omxplayer is being deprecated and resources are directed at improving vlc.
> 
> This is due to: omxplayer uses openvg for OSD and subtitles which isn't supported on Pi4. omxplayer uses openmax which has been deprecated for a long time and isn't supported with 64-bit kernels. omxplayer does not support software decode omxplayer does not support advanced subtitles omxplayer does not support playback from ISO files. omxplayer does not integrate with the X desktop
> 
> Please try using vlc. If there are features of omxplayer that vlc does not handle then try reporting here.

I've tried running omxplayer with subtitles on my Pi 4 and it errors out. So it doesn't seem like it is possible, unless you have some more knowledge that I haven't found yet...?

KenT2, 2020-06-29:
I agree, I should have said Buster on Pi3 if you do not want 2 screens

najoshi, 2020-06-29:
Yeah, that's what I'm going to try next.... or I might try to write my own pp_vlcdriver.py. :)

KenT2, 2020-06-30:
If you want my proof of concept code you are welcome. Github Issue on development of VLC is here. There is still a need to solve the X window related black background problem, any thoughts

[https://github.com/popcornmix/omxplayer/issues/778#issuecomment-651388998](url)

najoshi, 2020-07-01:
I would definitely like to see your proof-of-concept code? How do I get it?
- Nik.

KenT2, 2020-07-01:
Code attached.
You need to install python-vlc
      sudo pip3 install python-vlc

You can get the media from pp_home, copy it to the directory with the code

Type 1 to 5 in the window, some of the videos appear on HDMI-2

[vlcpy.py.txt](https://github.com/KenT2/pipresents-gapless/files/4858368/vlcpy.py.txt)

KenT2, 2021-07-03:
A VLC based videoplayer is now in pipresents-beep. Subtitles work

closed 2021-07-03 by KenT2