KenT2/pipresents-gapless #134: OSC control

DanielPineiroCalvo, 2020-08-02:
Hello Ken!v
First of all I want to thank you for developing this wonderful program. I am trying to use Pipresents to launch videos for a theater play, using OSC. I have some experience working with OSC (communicating programs like Isadora, Gobutton, etc.) but I am really stuck with Pipresents. I can send messages both with OSCremote (local) and also with Isadora or Qlab running on another computer.  
I have checked the IPs, the ports and I can see the messages received in the terminal window, but Pipresents does not respond to them.

I have reviewed the manual, but the example pp_osc_1p3 (for control by OSC) is not found in the directory, so I cannot get an idea of ​​how it works.

I would be very grateful if you could help me on this!!

Thanks and regards

Daniel

KenT2, 2020-08-03:
Sounds like you have not installed the examples,  pp_osc_1p3 is on Github . Instructions in the pipresents-gapless repository

DanielPineiroCalvo, 2020-08-04:
Ok! I´ll have a look on it. Thanks!!!

DanielPineiroCalvo, 2020-08-08:
Hi! I installed pp_osc_1p3. Now I can see the osc messages coming from pipresents in pp_oscremote.py (I configured all on localhost). I can see as well the messages that I send from pp_oscremote.py (also on locahost). But the messages don not take any efect.
When launching pp_oscremote.py I got the following message in terminal:

Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1550, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 599, in callit
    func(*args)
  File "pp_oscremote.py", line 127, in run_app
    self.output_reply_server=self.init_server(self.osc_config.this_unit_ip,self.osc_config.reply_listen_port,self.output_client)
  File "pp_oscremote.py", line 264, in init_server
    return OSC.OSCServer((ip,int(port_text)),client)
  File "/home/pi/pipresents/OSC_plus.py", line 1362, in __init__
    UDPServer.__init__(self, server_address, self.RequestHandlerClass)
  File "/usr/lib/python2.7/SocketServer.py", line 420, in __init__
    self.server_bind()
  File "/usr/lib/python2.7/SocketServer.py", line 434, in server_bind
    self.socket.bind(self.server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 98] Address already in use

KenT2, 2020-08-09:
I have not used PP and OSC for some years so this answer is a bit of a guess.

The error is because you are running two programs on the same Pi both using the same port. I think you need to run Pi Presents and pp_oscremote on different PI's

If you run with the -d command line option PP will indicate when it has recived an OSC message and the generated event.

KenT2, 2021-07-04:
Hopefully solved

closed 2021-07-04 by KenT2