KenT2/pipresents-gapless #135: Video Player error

wwagonman, 2020-10-14:
I have a large number of pi zeros running at a non profit museum.  Historically they have performed well.  Recently, however, I am getting random failures on various pi's where I get the following error:

Profile Error: Track file not found /media/pi/WDHM/Video.mp4

I have the pi's configured to run the video in a continuous loop.  The video is on a USB flash drive "WDHM".

I have experimented with changing out (one at a time) the pi's, the SD cards and the USB drives.  No combination fixes the issues.  Sometimes they run for a week without a problem, sometimes only overnight.  They do recover when rebooted.

Thanks for any help.

Lee

KenT2, 2020-10-17:
Hi
The error message is produced when PP checks that the file exists on the disk. A very simple operation which, unlike the actual playing of the video, should never cause problems.

Has anything else changed before the problem started to occur. An update to Raspbian. Maybe even the mains power or the introduction of a new electrical item that may cause interference, also intermittent power outages (just light flickering).

As and experiment it would be worth eliminating the USB connection by putting the media on the SD card. 

A three line python program would test this out, something like this
`import os, time
while True:
    if not os.path.exists('/media/pi/WDHM/Video.mp4'):
        print 'failed'
    time.sleep(1)`

wwagonman, 2020-10-17:
Thanks for the reply Ken.  We did have a couple of issues over the last year.  In one, a series of power glitches took out some of the SD cards.  Then with Covid, we changed a number of the displays to run in a loop rather than triggered by a button push.  At that time of the power glitches, I created new SD cards with the latest OS release and "gapless" for a new image.

For what it is worth, I set up a brand new Pi Zero with a fresh SD card and USB drive with the video on it using the same image that I have at the museum.  It would consistently fail within 24 hours.  I then took the SD card and USB drive and put them on an old Raspberry Pi 2 version B V1.1 that I had kicking around.  It has been running without fail for a week now.  Maybe the zero can't handle the new OS with PiPresents consistently?

Will try the experiment you suggested as well.

Thanks for the help.

Lee

KenT2, 2020-10-17:
Thanks for the update. I am beginning to suspect you are running out of RAM.

If you installed a later operating system after  the power glitch problem then the later version will probaby use more RAM and you might be running out. The various randomnesses in Raspbian will make this only happen occasionally.

This might explain the difference between the zero and 2B as the 2B has 1GB against 512MB for zero.

I notice I recommend 256MB of GPUMEM. This is well over the top for your application. 128MB for GPU  should be enough. 

Suggest you try this rather than my earlier ideas.

wwagonman, 2020-10-17:
Thanks Ken,  I have changed the GPUMEM to 128MB and restarted the zero that
was consistently failing.  Will report back.

Regards,

Lee

wwagonman, 2020-11-01:
Hi Ken,  wanted to report back.  Looks like the GPU memory was the trick.  The two pi's that I changed the GPU memory on have been chugging right along.  Thanks for your help and a great product!

KenT2, 2020-11-03:
That's good. I need to add something to the manual. What GPUMEM are you now using

wwagonman, 2020-11-04:
I used 128.  Thanks!

KenT2, 2021-07-04:
Added to pipresents-beep manual

closed 2021-07-04 by KenT2