KenT2/pipresents-gapless #136: Question: Startup time of pipresents

masseelch, 2021-04-13:
I experience (a small) lag when starting PiPresents. It takes about 4 to 5 seconds before the show actually starts on screen. What is causing such a long delay before the first track is shown on the screen? Can this be fastened up by changing some configuration options?

```json
// pp_showlist.json
{
 "issue": "1.4.4",
 "shows": [
  {
   "admin-colour": "#ffffff",
   "admin-font": "Helvetica 10 bold",
   "admin-justify": "left",
   "admin-x": "100",
   "admin-y": "200",
   "audio-speaker": "stereo",
   "background-colour": "",
   "background-image": "",
   "controls": "pp-down down\npp-stop stop\npp-pause pause",
   "disable-controls": "no",
   "disable-show-control-events": "no",
   "display-name": "HDMI0",
   "duration": "10",
   "eggtimer-colour": "#ffffff",
   "eggtimer-font": "Helvetica 10 bold",
   "eggtimer-justify": "left",
   "eggtimer-text": "Loading....",
   "eggtimer-x": "100",
   "eggtimer-y": "100",
   "empty-text": "Nothing to show",
   "enable-catchup": "yes",
   "freeze-at-end": "yes",
   "freeze-at-start": "no",
   "image-rotate": "0",
   "image-window": "original",
   "live-tracks-dir1": "",
   "live-tracks-dir2": "/home/pi/pp_home/live_tracks",
   "medialist": "artliveshow.json",
   "mplayer-audio": "hdmi",
   "mplayer-other-options": "",
   "mplayer-volume": "0",
   "omx-audio": "hdmi",
   "omx-other-options": "",
   "omx-volume": "0",
   "omx-window": "warp",
   "pause-timeout": "",
   "repeat": "repeat",
   "sched-everyday": "",
   "sched-monthday": "",
   "sched-specialday": "",
   "sched-weekday": "",
   "sequence": "ordered",
   "show-canvas": "",
   "show-control-begin": "",
   "show-control-end": "",
   "show-control-events": "",
   "show-html-background-colour": "#ffffff",
   "show-html-height": "300",
   "show-html-width": "300",
   "show-ref": "artliveshow",
   "show-text": "",
   "show-text-colour": "#ffffff",
   "show-text-font": "Helvetica 20 bold",
   "show-text-justify": "left",
   "show-text-location": "",
   "show-text-type": "plain",
   "show-text-x": "0",
   "show-text-y": "0",
   "title": "artliveshow",
   "track-text-colour": "#ffffff",
   "track-text-font": "Helvetica 20 bold",
   "track-text-justify": "left",
   "track-text-x": "0",
   "track-text-y": "40",
   "transition": "cut",
   "type": "artliveshow"
  },
  {
   "background-colour": "#000000",
   "counters-initial": "",
   "counters-store": "no",
   "sched-enable": "no",
   "sched-everyday": "",
   "sched-monthday": "",
   "sched-specialday": "",
   "sched-weekday": "",
   "show-ref": "start",
   "sim-day": "",
   "sim-hour": "",
   "sim-minute": "",
   "sim-month": "",
   "sim-second": "",
   "sim-year": "",
   "simulate-time": "no",
   "start-show": "artliveshow",
   "start-show-ref": "start",
   "title": "Start",
   "type": "start"
  }
 ]
}

// artliveshow.json
{
 "issue": "1.4.4",
 "tracks": []
}
```

There is only one image in the live_tracks dir.

KenT2, 2021-04-13:
Are you connected to a network?

PP waits for a few seconds for a network to be available. You can disable the wait with the -nonetwork command line option.

masseelch, 2021-04-14:
`--nonetwork` was the correct command. However, there still is the little lag when starting the show. I think it has something to do with the size of the image I am showing. The smaller it is, the faster PP loads. It seems like PNG-images are slower than JPG-images.

KenT2, 2021-04-15:
There will always be a small lag before the first track when the image/video loads. For all tracks after that the loading is done while the previous track is playing.
Images are loaded and displayed using PIL (pillow). Make sure the image size is no larger than you need it to be to speed things up

masseelch, 2021-04-15:
I downsized the images to my screen resolution and did the same with the vidoe and now there is super smooth running. Very much thanks :-)

closed 2021-04-15 by masseelch