KenT2/pipresents-gapless #137: How to start the show by clicking on a link on a locally stored webpage?

sky-love, 2021-05-09:
This may be a silly question but I can't find any info how to start the show by simply clicking on a link on locally hosted website? I am trying with PHP with "echo shell_exec('python /home/pi/pipresents/pipresents.py -p Art -f -b') to no avail. The above show is working flawlessly when I type this command straight into the terminal but I need to trigger it via a html link or a button. Thank you!

KenT2, 2021-05-09:
You may need to specify the complete path of the profile and python e.g.

/usr/bin/python /home/pi/pipresents/pipresents.py -o /home/pi -p myprofile

See Section 6.6 of the manual where autostart has a similar solution

sky-love, 2021-05-10:
I am trying another approach and using your Manager (PP_manager.py) to see if I can control opening my slideshow this way. The manager starts but it gives me the error when I browse to the specified IP. Please see below.. Thank you for your help!

 *** Pi Presents Manager Started ***
Manager: Found pp_manager.py in  /home/pi/pipresents
Manager: Waiting for Network
Manager: Found pp_web.cfg in  /home/pi/pipresents/pp_config/pp_web.cfg
Manager: Network details My Pi wlan0 192.168.0.102
Manager: Found pp_email.cfg at /home/pi/pipresents/pp_config/pp_email.cfg
remi.server      INFO     Started websocket server 192.168.0.102:33591
remi.server      INFO     Started httpserver http://192.168.0.102:8081/
remi.server.http INFO     built UI (path=/)
Manager: Network Details wlan0 192.168.0.102
Manager: read email options from /home/pi/pipresents/pp_config/pp_email.cfg
Manager: Web server started by pp_manager
remi.server.http ERROR    error processing GET request
Traceback (most recent call last):
  File "/home/pi/pipresents/remi/server.py", line 789, in do_GET
    self.client.root = self.main(*self.server.userdata)
  File "pp_manager.py", line 259, in main
    self.display_state()
  File "pp_manager.py", line 793, in display_state
    pid,user,profile=self.pp.is_pp_running()
  File "pp_manager.py", line 1465, in is_pp_running
    p_index=split.index('-p')
ValueError: '-p' is not in list
remi.server.http INFO     built UI (path=/)
Manager: Network Details wlan0 192.168.0.102
Manager: read email options from /home/pi/pipresents/pp_config/pp_email.cfg
Manager: Web server started by pp_manager
remi.server.http ERROR    error processing GET request
Traceback (most recent call last):
  File "/home/pi/pipresents/remi/server.py", line 789, in do_GET
    self.client.root = self.main(*self.server.userdata)
  File "pp_manager.py", line 259, in main
    self.display_state()
  File "pp_manager.py", line 793, in display_state
    pid,user,profile=self.pp.is_pp_running()
  File "pp_manager.py", line 1465, in is_pp_running
    p_index=split.index('-p')
ValueError: '-p' is not in list

KenT2, 2021-05-11:
pp_manager is using the ps -A -o cmd  command  to see whether there is a copy of PP running to update its status.The running copy does not have to be started by pp_manager. It looks like there is a copy of pipresents running that has been started without specifying the -p option. PP_manager is not forgiving in this area.
If you start the Task Manager from Accessories>Task Manager and select the View Full Command Line option you will see what commands are currently being  executed.