KenT2/pipresents-gapless #139: Linking together Pipresents Radiobutton show

Giles4604, 2021-07-07:
Hi Ken,

Would you be able to give me some guidance on how to link Radiobutton shows together? What I am hoping to do is set up a front page displaying the alphabet. By clicking on a letter this should open up another page of buttons which is a list surnames. Then by clicking on a surname open up a further page with an image of that person and a return to menu button.

Is it possible to set this up using pipresents?

 Many thanks 

Giles

KenT2, 2021-07-09:
It is possible, as with any show a track may be a show. However I would recommend a hyperlinkshow rather than a radiobuttonshow. The hyperlinkshow has a better command set for what you want.

A radiobuttonshow has one special feature, you can use controls to start another track while one is playing. To do this input events are not passed down to subshows which is not what you want. You can disable this feature in the profile but if you do you might as well use a hyperlinkshow.

Giles4604, 2021-07-12:
Hi Ken,

Thank you so much for your advice. I will try this out using a hyperlinkshow.