KenT2/pipresents-gapless #140: Struggling with LED on GPIO output

StemsMusicMan, 2021-09-09:
Hi Ken,

Thanks for your amazing work.  I've got an exciting project that will be going live in a couple of weeks fingers crossed that we're using Pi Presents for.  I can fill you it on that very soon!

In the meantime, I'm struggling with getting an LED to light and I can't see what I'm doing wrong.  I've set up a GPIO.cfg within my profile's folder.  All the inputs (basically all switches) are operating correctly.

I want to switch on an LED as soon as the profile runs (it's set to run on boot up).  In my GPIO.cfg I've specified pin 23 like this (though I have tried others to troubleshoot):

[P1-23]
direction = out
name = blueled

Within my menushow.json file right at the top I've got:

{
 "issue": "1.3.5",
 "tracks":  [
   {
    "animate-begin": "0 blueled state on",
    "animate-clear": "no",
    "animate-end": "",

and then background etc and all my media listed underneath.....

My understanding is that when the profile is ran, the animate-begin function will make the 'blueled' logical name turn 'on', and thereby create +3.3v at all the pins in GPIO.cfg named blueled, i.e. pin 23.

I then want the LED to remain lit until the Pi is shutdown, hence why the animate-clear function is still 'no'.

I think I might have missed something.  Is there an extra step or have I got it all wrong.  Any help much appreciated!

All the best,
Steve

KenT2, 2021-09-10:
Glad it is fixed.
I may be wrong but the way you have done it the LED will turn on only when one of the tracks in the menu is started.

If you want the LED to be on when the profille is run then I would create a second show which is put in Start Shows so it runs at the start. This mediashow will have a single audio track with a blank location and infinite duration. Put the animate command in this track. I designed this special variant of an audio track so it could do things like animation.

StemsMusicMan, 2021-09-21:
Thanks Ken, the LED is working as I'd hoped; it illuminates as soon as the profile runs, and stays on until I press a switch enabling the 'pp-shutdownnow' command.

Another question if I may, are you aware of a way that mp3s can play without a delay?  I trigger the start of an mp3 with a remote signal going into the GPIO, and that signal is sent to multiple Pis at the same time.  Sometimes all of the Pis start exactly at the same time, but quite often there are a few milliseconds delay on one or two of the Pis so they are playing back slightly out of sync.  I did wonder if there's a way that the track can start and immediately pause, waiting for a trigger to effectively restart the track simultaneously across all the Pis.  Would you have any suggestions?  Cheers for the help so far, Steve

KenT2, 2021-09-24:
I suspect there is no answer to this as Raspbian is not a real time operating system and there will always be random delays.
You could try the video playout which is meant to be for video tracks. In this PP pauses just before the first frame and a 'g' command unpauses it. omxplayer plays mp3 audio tracks but I am not sure if it will pause before the first frame as there are no frames.

The other possibility is to move to Pi Presents Beep and use the beep facility. This facility plays a mp3 track triggered directly from an event using mpg123 which is a lightweight audio player so loads quickly.