KenT2/pipresents-gapless #142: Photo Eye with PiPresents

luluh79, 2021-09-23:
I am using a photo eye sensor with PiPresents. It was working fine, but now after triggering the show and a script once, it repeatedly continues triggering. Any idea what might be causing this?

KenT2, 2021-09-24:
Is it possible that you have edited gpio.cfg so that the event is based on the state rather than an edge or you have made it a repeating event. This seems most likely if the repeating trigger only happens after the first.

Otherwise it sounds a bit like electrical noise causing false triggering. The -d command line option will show a log including detected events. Also there is a programme input_test.py in the pipresents folder that you can use to display the state of the pins.

KC9YIA, 2021-09-26:
We are using the enforcer Retro-Reflective Sensors E-931-S-35RRQ wired for normally closed.  We have groups of 2+ people come thru and every person in the group breaks the beam causing PiPresents to toggle from the ambient track to the triggered track until the last person is thru the beam. Is there a way to make PiPresents ignore all but the first trigger change? 

GPIO.cfg text

[P1-07]
direction = in
rising-name = 
falling-name = trigger
one-name = 
zero-name = 
repeat = 
threshold = 1
pull-up-down = up

The falling name matches in the input parameters in the web editor

Also when I run the -d command I see that when triggered that OMX player is being killed completely and then reloading a new instance  of OMX player causing the screen to go black and not giving us a seamless transition between the two tracks

KenT2, 2021-09-28:
I assume you are using a radiobuttonshow and in the show have a control like 'trigger play mytrack'
To stop the track being played a second time put this control in the track
'trigger null'

The other thing that might help is to increase the threshold in gpio.cfg.

You will not be able to hide the gap completely. To reduce the gap ensure that in the first video track freeze at end is set to yes. A background colour other than black helps.

KC9YIA, 2021-09-28:
We are using the media show. We need the first track to loop until triggered and then switch to the second track.

KenT2, 2021-09-28:
I don't think mediashow will work because the 'next-trigger' will move to the next video each time somebody breaks the beam.

Just tried a radiobuttonshow with a video as first track - the video loops

If you add one other video track, track reference track1,  and add 'trigger play track1' to the show, also 'trigger null' to track1 it should hopefully do what you want and not retrigger the video until it has finished.