KenT2/pipresents-gapless #143: Struggeling with autostart up

FrozoneAirPol, 2021-09-29:
Hello, 

Thanks for the awesome program. 

I have raspbian Buster installed (10) 

I'm trying to set up the autostart, but it's not working.

Steps followed 
`this will function only if you have set 'boot to desktop' using raspi-config.`
Done

`The directory .config is already present in the image but you will need to select 'Show Hidden Files' in the File Manager to see it.
`
Found

`For later Stretch versions the autostart file:/home/pi/.config/lxsession/LXDE-pi/autostartdoes not exist so it, or its contents, needs to be copied from the file:/etc/xdg/lxsession/LXDE-pi/autostart then edit the file/home/pi/.config/lxsession/LXDE-pi/autostart`
If I'm getting it right: I went to the /etc/xdg/lxsession/LXDE-pi/autostart file, copied it to /home/pi/.config/lxsession/LXDE-pi/ that I created myself

`Add the following line below the last line to start Pi Presents with the required options:/usr/bin/python /home/pi/pipresents/pipresents.py -o /home/pi -p myprofileNote: the file is not a standard .sh file. The leading @ is not required, it means restart the process should it fail.`

Added a functional command-line in the at the bottom of the copied autostart file in /home/pi/.config/lxsession/LXDE-pi/

Nothing is happening at start up, looks normal to me that the system is not looking for that autostart file, because I never told him to check that file.

Thanks in advance if you have some ideas for me.

KenT2, 2021-09-30:
> Nothing is happening at start up, looks normal to me that the system is not looking for that autostart file, because I never told him to check that file.
It's not that. RPi OS will use the autostart file if it is there.

Just checking a couple of points:
- Does the functional command line in the autostart file have the full paths of all the files as in the instructions?.
- Have you changed the profile to from myprofile to yours?

If you could attach the autostart file that would be good.

FrozoneAirPol, 2021-09-30:
Hi Ken, 

so the file autostart in /home/pi/.config/lxsession/LXDE-pi contains
`@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
/usr/bin/python /home/pi/pipresents/pipresents.py -o /home/pi -p video_loop -f `

And the file autostart in /etc/xdg/lxsession/LXDE-pi
`@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
`

The command line 
`/usr/bin/python /home/pi/pipresents/pipresents.py -o /home/pi -p video_loop -f `
works when used in the terminal. With video_loop being a folder in /home/pi/pp_home/pp_profiles/video_loop

Thanks,

FrozoneAirPol, 2021-09-30:
Hi Ken, 

First there were still some errors my profile, so good to know the validate option. 

After solving the errors, tried again and didn't worked. 

Then I ran the debug and now it's working with the same command line and now it's working. 

Thanks a lot