KenT2/pipresents-gapless #144: [Errno 13] Permission denied: '/home/pi/pipresents/pp_logs/pp_log.txt' Why??

atthedropofahat, 2021-11-30:
Apologies if this is simple, however I cannot find the same issue elsewhere

python /home/pi/pipresents/pipresents.py
Traceback (most recent call last):
  File "/home/pi/pipresents/pipresents.py", line 892, in <module>
    pp = PiPresents()
  File "/home/pi/pipresents/pipresents.py", line 89, in __init__
    self.mon.init()
  File "/home/pi/pipresents/pp_utils.py", line 192, in init
    Monitor.ofile=open(Monitor.log_path+ os.sep+'pp_logs' + os.sep + 'pp_log.txt','w',bufsize)
IOError: [Errno 13] Permission denied: '/home/pi/pipresents/pp_logs/pp_log.txt'

This is on a fresh RPi Buster image all packages needed were installed updated and upgraded. Only thing different to my previous install is a USB RTL-SDR and rtl_433 install

atthedropofahat, 2021-11-30:
My bash script for install (excludes the directory copy steps which I perform manually)

#!/bin/bash

# UPDATE and UPGRADE
apt-get update
apt-get upgrade -y
# INSTALL PYTHON3
apt-get install python3 -y
# INSTALL PIPRENSENTS Supporting packages
apt-get install python-imaging -y
apt-get install python-imaging-tk -y
apt-get install unclutter -y
apt-get install mplayer -y
apt-get install uzbl -y
apt-get install python-pexpect -y
pip install evdev -y
# START VNC
apt-get install real vnc-server -y
vncserver
# UPDATE and UPGRADE
apt-get update
apt-get upgrade -y
#GET PIPRESENTS
cd /home
wget https://github.com/KenT2/pipresents-gapless/tarball/master -O - | tar xz
wget https://github.com/KenT2/pipresents-gapless-examples/tarball/master -O - | tar xz

KenT2, 2021-12-01:
Bit if a mystery this one.

I can't see anything wrong with what you are doing except for the unnecessary install of python3 and vncserver in the bash script.

The error message indicates that the file /home/pi/pipresents/pp_logs/pp_log.txt or its directory, or maybe the /pipresents directory has been made read only. You can check them in the file explorer (right click). 

Your only difference to what I do is that I download the zip file from github and copy manually, and do not use a bash script at all. Others use the wget and I've not had complaints.

One slight oddity in the script sudo apt install real vncserver -y   Should there be a space between real and vncserver.

atthedropofahat, 2021-12-01:
> Bit if a mystery this one.
>
> I can't see anything wrong with what you are doing except for the unnecessary install of python3 and vncserver in the bash script.
>
> The error message indicates that the [...]

Hi Ken. I made a bash after my first install attempt failed, seemed sensible for making a quicker install process. Same failure both times using the wget.

I had PiPresents working on another RPi previously without issue using wget. Thanks for pointing out the space between real and vnc. I will change this.

It looks like the whole pipresents folder is only read only other than root. Will let you know the outcome.

atthedropofahat, 2021-12-01:
Hi Ken,

For some reason (not entirely sure why using either script caused this) the owner was root and the permissions were only for root to write. I tried chmod withour success but 

sudo chown pi /home/pi/pipresents -R
sudo chown pi /home/pi/pp_home -R

did the trick.

KenT2, 2021-12-01:
did a quick google. shell scripts are run as root for some peculiar reason. Need to use su  to fix it - 

[https://stackoverflow.com/questions/21578852/running-shell-script-after-boot-on-raspberry-pi](url)