KenT2/pipresents-gapless #145: Pi-Presents Gapless OSC command not working with spaces

morte615, 2021-12-10:
Using Gapless and I'm trying to send OSC commands from my computer to Pi-Presents. I have it working when I use a OSC command that is a single line with no spaces (ex. /pipresents/DoorLock/core/reboot) but as soon as I try to use an event, or any command that requires a space before the parameters, it won't work (ex. /pipresents/DoorLock/core/event pp-stop) and returns this error in the pi-presents window:

> MESSAGE RECEIVED
> from:  ('xxx.xxx.xxx.xxx', 5882)
> raw contents;  /pipresents/DoorLock/core/event pp-stop, 
> 
> 839.01  Warning:  OSCDriver_3027054544:  No handler for message from xxx.lan:5882
>      /pipresents/DoorLock/core/event pp-stop

I'm getting the idea that it's not declared somewhere as even if I put nonsense in it gives me the exact same error. Everything works fine on the Pi itself just the OSC command not working. Is it the space that is causing the issue or do I need to declare pp-stop (and pp-key-0 through pp-key-8) somewhere?

morte615, 2021-12-10:
As an aside I tried adding it directly to the keys.cfg file with no change. Though since it works without it being there I'm assuming that's not where it needs to be added.

morte615, 2021-12-10:
So used a different OSC generator and I was able to control it. So it's something on how I'm formatting the command. I'll look deeper at my stuff. Thanks!

KenT2, 2021-12-10:
The error message is being produced by the OSC module not Pi Presents. I suspect the space is not a space.

To use just associate pp-stop with a command in the controls field of a track or show. pp-stop is the symbolic name of an input event.