KenT2/pipresents-gapless #146: Current installation manual still up to date?

maxholgasson, 2021-12-18:
Hi,
when installing this gapless version on a duster system according to the instruction I'll receive the error 
E: Für Paket »python-imaging« existiert kein Installationskandidat.

What is the correct way and can you update the manual?

lxne, 2021-12-22:
Same here on buster.

KenT2, 2021-12-22:
Thanks for report. 

Google's translation is 'python-imaging «there is no installation candidate'. Is the translation correct?

Did Pi Presents work OK if you display an image even though RPi OS reports something missing.

It worked OK when I created a new Buster legacy image a few days ago. I'll look into it.

lxne, 2021-12-22:
Al the demos seem to work OK.
I am new to Pi Presents but everything seems to work fine.

KenT2, 2021-12-22:
That's good. I think the problem is explained here but need to check this:

https://github.com/KenT2/pipresents-gapless/pull/141

Python imaging has been replaced by PIL so you are trying to install an unnecessary package and it replacement is already installled in the downloaded image.

maxholgasson, 2021-12-23:
Hi, thanks yes it worked with PIL however I couldnt get it to run playing videos with audio over hdmi..I set up the raspberry completly new and then installed Pi presents 1.4 which works fine now. (even if I don't get rid of the gap between videos, freeze at start freezes the whole video right from the beginning, freeze at start doesn't help either..but that's for another issue)