KenT2/pipresents-gapless #147: Sending OSC commands to a local listener

LevitatingBusinessMan, 2022-01-24:
Currently it doesn't really seem possible for PiPresents to send OSC commands to itself.

I have a Raspberry Pi that is running pipresents, and also listens to OSC commands (on a seperate port than PiPresents) to control a Music Player Daemon server.

I want PiPresents to be able to control this Music Player Daemon by sending OSC commands to localhost. But this doesn't seem possible because PiPresents listens to the same port it sends OSC commands to (to listen for a reply).

KenT2, 2022-01-25:
Its been a long time since I implemented OSC but I wonder if PP and the MPD could use the same port. The /pipresents field of the message being used to filter out the MPD messages.

Otherwise could you use 2 PI's or use PP to play the sounds.

LevitatingBusinessMan, 2022-01-29:
@KenT2 Thanks for your response Ken, I have decided to split the project into two Pi's. Much love to you and your project!

Cheers

closed 2022-01-29 by LevitatingBusinessMan