KenT2/pipresents-gapless #151: Question about a show based on magicpicture

Knightrider6224, 2022-04-11:
Hi, 
Sorry to post this question but I'm struggling with a small addon I'd like to make to a show based on magicpicture example.
This show runs perfectly for more than two years now, each video ( 8 playing in random ) is triggered by a PIR sensor or a RF remote control both connected to the same GPIO, I've added a mechanical switch to jump from the sensor function to the remote. 
I'm trying now for 3 days to add a second button of the RF remote ( or pir sensor, connections are the same ), this second button should launch a specific video file and only this one.
I've added a new binding in the GPIO config file but I don't know how to tell this video file to respond to it.
Could you please give me a hint ? that could lead me on track.

Many thanks

KenT2, 2022-04-12:
Hi
You need to add a child track to the mediashow. The Play command will then play the video and return to the magic picture show. Hopefully this will interrupt the Freeze at Start but I'm not sure about this.

KenT2, 2022-04-13:
Yes that is correct. pp_interactive_1p3 is an example of a child. It is slightly different as the child is a menushow not a video track