KenT2/pipresents-gapless #153: hyperlink show touchscreen screen saver issue

eehindman, 2022-09-12:
Hi, I'm using a hyperlink show with a touch screen. I have a video as a screen saver and made the whole screen one big touch button to return to the home page. This has been working just fine with a video that was shot from a phone and used as a place holder until I could make the screen saver I wanted. It goes to screen saver and then when touched returns to home page just like it's supposed to.

But, now that I've made the actual screen saver, when I replace it, it works the first time and then never again. It goes to screen saver and then when touched goes to home screen but after it goes to screen saver a second time, it will no longer react to touch. When I change it back to the place holder video, everything works as it should.

The place holder video is a 65MB mp4
the "real" video is a 23MB mp4
they are both 60 seconds long

I have tried another video from a phone and it doesn't have the issue. The "real" screen saver I made in after effects and rendered in H.264. I don't know what's wrong. If anyone can help I would be very, very grateful.
Thanks!

KenT2, 2022-09-13:
The Raspberry Pi has always been a bit temperamental playing videos. Some suggestions:
Have yet set GPU_MEM as in the Github instructions
Is the video full HD. Pi does not do well with fullHD (1920*1080) try 720p instead.
If these fail use the -d command line option and post the debug log in pipresents/logs/log.txt

eehindman, 2022-09-17:
> 

The 720p worked. Thank you very much!

closed 2022-09-17 by KenT2