KenT2/pipresents-gapless #154: Execute OS process

wdonker, 2023-04-20:
More a question than an issue:
I'm using PiPresents Gapless and want to switch a Wifi smart powerplug when starting or ending a show. The plug can be switched by http. Is there a (easy) way to send the http-command from PiPresents (maybe using curl)?

KenT2, 2023-04-20:
Not easy, you will need to write a little python code. Either

1. Use an I/O plugin. Using this an output event will send the http: command. There are instructions in the manual for writing the plugins.
2. add a Show Control command. The code for the current commands is in pipresents.py at around line 546. Probably best to modify the code for an existing command.

2. is probably easier. Whichever you do the code should not block for more than 100mS as PP uses cooperative scheduling.

wdonker, 2023-04-21:
Thank you for the suggestions, Ken. I will see what is a feasible solution for me. The I/O plugin seems the most clean but maybe also the most difficult to implement.
For now, this issue can be closed.